CRLG/GROSBOT_MBED #1: Gestion des autotests au démarrage MBED

CRLG, 2017-08-08:
Gérer un autotest au démarrage
Tester tous les capteurs au démarrage : présence sur l'I2C / Valeur cohérente
Tester la communication avec chaque dsPIC
Tester les actionneurs : 
	faire bouger la roue gauche vers l'avant et vérifier que le codeur gauche a été en avant
	faire bouger la roue gauche vers l'arrière et vérifier que le codeur gauche a été en arrière
	Idem avec la roue droite
Prévoir une trame vers LABOTBOX pour remonter un flag par équipement
Prévoir un paramètre EEPROM pour inhiber/activer l'autotest au démarrage

laguiche, 2017-08-08:
@CRLG voire également :
Catégoriser les défauts et pouvoir calibrer le bounce et le debounce. 😉

_Sent from my WIKO U FEEL PRIME using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

NicoFarmer, 2019-03-28:
Précision sur les autotest à réaliser : 
   - Tension batterie dans une plage donnée
   - Présence de chaque dsPIC n°1 et n°2 sur le bus I2C
   - Présence de chaque capteur ultrason sur le bus I2C
   - Présence du SD20 pour le pilotage des servos sur le bus I2C
   - Présence de la carte PowerSwitch sur le bus I2C
   - Pour chaque moteur de traction / codeur : faire bouger chaque roue et vérifier le retour d'état sur le codeur
   - Pour l'ascenseur : faire bouger le moteur et vérifier le retour d'état du codeur associé + vérifier les switchs de fin de course