CRLG/GROSBOT_MBED #30: Remplacement stratégie Yakindu

laguiche, 2019-11-07:
Malgré l'apparente facilité de design via l'ihm et de codage automatique,  l'utilisation de Yakindu a montré quelques écueils lors de la coupe 2019:
Travail collaboratif sur la stratégie (difficile, merge,...)
Temps de mise en œuvre 
Reprise de stratégie prototype
licence à renouveler chaque année 

Un remplacement de Yakindu par une autre solution permettrait donc de gagner du temps à la coupe pour se concentrer sur d'autres tâches et minimiser les essais et les erreurs.

NicoFarmer, 2019-12-28:
Suite à la réunion du club du 21/12/2019
Pourquoi on veut changer : 
  - Sponsoring à renouveler chaque année
  - Pas de licence pour tout le monde car licence nominative
  - Merge difficile entre plusieurs modèles 
  - Navigation dans le modèle longue
  - Pour retoucher un petit bout de modèle juste avant le match, c'est long
  - Obligation de re-coder en graphique la mise au point d'une séquence faite avec ActuatorSequencer

Décision : on part sur le proto Yakindu like Nico + ActuatorSeqencer de Stéph pour la génération de code
Présentation de la proposition technique du 21/12/2019
[19-12-2019-Yakindu_Like-Nico.pdf](https://github.com/CRLG/GROSBOT_MBED/files/4007254/19-12-2019-Yakindu_Like-Nico.pdf)

NicoFarmer, 2019-12-29:
L'architecture proposée en remplacement de Yakindu est nommée **Modelia** pour _Model_ et _IA_

NicoFarmer, 2022-05-11:
Structure "**Modelia**" mise en place : ticket soldé

closed 2022-05-11 by NicoFarmer